kochedykov/jlibmodbus #2: maven support

WolfgangFahl, 2017-07-19:
Vladislav - your library looks promising. I'd love to use it. 
It would be great if it would be "mavenized" and made available on maven central.

kochedykov, 2017-07-24:
Hello!

I think the same way. I'm going to add maven support after a little while.

Thanks for your feedback!

WolfgangFahl, 2017-07-28:
Thank you for the positive feedback. I would appreciate if you'd to the maven stuff first before you add more featurs. Currently I get a NoClassDefFound Error and its hard to work around this.

WolfgangFahl, 2017-07-28:
I am using my fork as a workaround for the time being

kochedykov, 2017-07-31:
Thank you, I've merged commits.

WolfgangFahl, 2017-07-31:
at
https://github.com/WolfgangFahl/Mediawiki-Japi
you can find an example of a github project that is available via maven central. 

here are pointers for some instructions

- http://central.sonatype.org/pages/ossrh-guide.html
- http://datumedge.blogspot.de/2012/05/publishing-from-github-to-maven-central.html
- http://central.sonatype.org/pages/apache-maven.html

kochedykov, 2017-07-31:
All preparations are done and I'm going to promote the first release today.

kochedykov, 2017-07-31:
Done. I've promoted the first release on oss.sonatype.org. 

<dependency>
  <groupId>com.github.kochedykov</groupId>
  <artifactId>jlibmodbus</artifactId>
  <version>1.2.8.4</version>
</dependency>

kochedykov, 2017-07-31:
Done. I've promoted the first release on oss.sonatype.org. 

`<dependency>
  <groupId>com.github.kochedykov</groupId>
  <artifactId>jlibmodbus</artifactId>
  <version>1.2.8.4</version>
</dependency>
`

PR #9 (2017-07-31): adds basic maven pom file
first step to fix #2
commit f2e0614d (2018-02-26): Fix #2 Error in JSerialComm implementation.
closed 2017-07-31 by kochedykov