kochedykov/jlibmodbus #3: Documentation and examples

WolfgangFahl, 2017-07-19:
Your library looks promising. I'd love to use. It would be great if you'd supply some simple examples. 
I am currently stuck with another  library at https://github.com/steveohara/j2mod/issues/52

kochedykov, 2017-07-24:
Hello!

I prepare documentation pages for this project. By now there is some documentation in Russian and I'm going to translate it to English.
Also, I can add a simple How-To document to the project's root.

<strike>Do not use j2mod, use jlibmodbus</strike> :-)

Thanks for your feedback!

gkrananth, 2018-02-15:
i have tried serial example to read float32 holding register value.  Please find the code 

              	ReadHoldingRegistersRequest request = new ReadHoldingRegistersRequest();
                    request.setServerAddress(slaveId);
                    request.setStartAddress(offset);
                    request.setQuantity(quantity);
                    ReadHoldingRegistersResponse response = (ReadHoldingRegistersResponse) request.getResponse();

                    m.processRequest(request);
                    ModbusHoldingRegisters registers = response.getHoldingRegisters();
                    for (int r : registers) {
                        System.out.println(r);
                    }
                    //get float
                    System.out.println("PI is approximately equal to " + registers.getFloat32At(0));

i am getting ILLEGAL Data_Value error code 3.  My register is floating 32 data type how do i get value for the register value of float32.  Integer is working.

kochedykov, 2018-02-15:
You get ILLEGAL Data_Value at line "m.processRequest(request);"?

gkrananth, 2018-02-15:
yes.

gkrananth, 2018-02-15:
appreciate if you can help me faster. Thanks

kochedykov, 2018-02-15:
At string "m.processRequest(request);" you just request multiple holding registers (16-bit values). If a device responds you with ILLEGAL_Data_Value see the documentation on the device (memory map). Perhaps, you made a mistake in address or quantity field.

kochedykov, 2018-02-15:
Note that reading float32 requires requesting 2 holding register.

gkrananth, 2018-02-16:
thanks for your reply, i am getting the following error even my quantity increased to 2

ipc01@ipc01-desktop:~/Downloads$ java -jar TestRTU.jar /dev/ttyUSB5 1 3110 2
INFO: Frame sent: 01030C2600022690
INFO: Frame recv: 0183030131
com.intelligt.modbus.jlibmodbus.exception.ModbusProtocolException: ILLEGAL_DATA_VALUE: Exception Code = 3
	at com.intelligt.modbus.jlibmodbus.ModbusMaster.processRequest(ModbusMaster.java:142)
	at com.hdb.ModbusTestRTU.main(ModbusTestRTU.java:99)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at org.eclipse.jdt.internal.jarinjarloader.JarRsrcLoader.main(JarRsrcLoader.java:58)

kochedykov, 2018-02-16:
You received the exception code 3 from the remote device with slaveId = 1, function code = 03. It means that the remote device cannot send you 2 valid register values with offset 0C26. You receive valid MODBUS EXCEPTION response from the remote slave and the library handle the response correctly. Can you show me the slave's memory-map?

gkrananth, 2018-02-16:
[image]

gkrananth, 2018-02-16:
> I do not see an element with offset 3110 (0x0C26).

Please find the attachment of modscan result. Is it can help you to
identify the issue?

gkrananth, 2018-02-16:
3110 is there but my attachment is referring the float32 register data type and even 2700 is also giving same error.

kochedykov, 2018-02-16:
please show the source code.

gkrananth, 2018-02-16:
package com.hdb;

import com.intelligt.modbus.jlibmodbus.Modbus;
import com.intelligt.modbus.jlibmodbus.ModbusMaster;
import com.intelligt.modbus.jlibmodbus.ModbusMasterFactory;
import com.intelligt.modbus.jlibmodbus.data.ModbusHoldingRegisters;
import com.intelligt.modbus.jlibmodbus.exception.ModbusIOException;
import com.intelligt.modbus.jlibmodbus.exception.ModbusNumberException;
import com.intelligt.modbus.jlibmodbus.exception.ModbusProtocolException;
import com.intelligt.modbus.jlibmodbus.msg.request.ReadHoldingRegistersRequest;
import com.intelligt.modbus.jlibmodbus.msg.response.ReadHoldingRegistersResponse;
import com.intelligt.modbus.jlibmodbus.serial.SerialParameters;
import com.intelligt.modbus.jlibmodbus.serial.SerialPort;
import com.intelligt.modbus.jlibmodbus.serial.SerialPortException;

import jssc.SerialPortList;

/*
 * Copyright (C) 2016 "Invertor" Factory", JSC
 * All rights reserved
 *
 * This file is part of JLibModbus.
 *
 * Redistribution and use in source and binary forms, with or without modification,
 * are permitted provided that the following conditions are met:
 *
 * 1. Redistributions of source code must retain the above copyright notice,
 * this list of conditions and the following disclaimer.
 *
 * 2. Redistributions in binary form must reproduce the above copyright notice,
 * this list of conditions and the following disclaimer in the documentation and/or
 * other materials provided with the distribution.
 *
 * 3. Neither the name of the copyright holder nor the names of its contributors may be used to endorse
 * or promote products derived from this software without specific prior written permission.
 *
 * THIS SOFTWARE IS PROVIDED BY THE COPYRIGHT HOLDERS AND CONTRIBUTORS "AS IS" AND ANY EXPRESS OR IMPLIED WARRANTIES,
 * INCLUDING, BUT NOT LIMITED TO, THE IMPLIED WARRANTIES OF MERCHANTABILITY AND FITNESS FOR A PARTICULAR PURPOSE ARE DISCLAIMED.
 * IN NO EVENT SHALL THE COPYRIGHT HOLDER OR CONTRIBUTORS BE LIABLE FOR ANY DIRECT, INDIRECT, INCIDENTAL, SPECIAL, EXEMPLARY,
 * OR CONSEQUENTIAL DAMAGES (INCLUDING, BUT NOT LIMITED TO, PROCUREMENT OF SUBSTITUTE GOODS OR SERVICES; LOSS OF USE, DATA, OR PROFITS;
 * OR BUSINESS INTERRUPTION) HOWEVER CAUSED AND ON ANY THEORY OF LIABILITY, WHETHER IN CONTRACT, STRICT LIABILITY,
 * OR TORT (INCLUDING NEGLIGENCE OR OTHERWISE) ARISING IN ANY WAY OUT OF THE USE OF THIS SOFTWARE,
 * EVEN IF ADVISED OF THE POSSIBILITY OF SUCH DAMAGE.
 *
 * Authors: Vladislav Y. Kochedykov, software engineer.
 * email: vladislav.kochedykov@gmail.com
 */
public class ModbusTestRTU {

    static public void main(String[] arg) throws SerialPortException {
        SerialParameters sp = new SerialParameters();
        Modbus.setLogLevel(Modbus.LogLevel.LEVEL_DEBUG);
        try {
            // you can use just string to get connection with remote slave,
            // but you can also get a list of all serial ports available at your system.
       //     String[] dev_list = SerialPortList.getPortNames();
            // if there is at least one serial port at your system
         //   if (dev_list.length > 0) {
                // you can choose the one of those you need
                sp.setDevice(arg[0]);
                // these parameters are set by default
                sp.setBaudRate(SerialPort.BaudRate.BAUD_RATE_19200);
                sp.setDataBits(8);
                sp.setParity(SerialPort.Parity.EVEN);
                sp.setStopBits(1);
                //you can choose the library to use.
                //the library uses jssc by default.
                //
                //first, you should set the factory that will be used by library to create an instance of SerialPort.
                //SerialUtils.setSerialPortFactory(new SerialPortFactoryRXTX());
[... 29 lines cut ...]
                    for (int r : registers) {
                        System.out.println(r);
                    }
                    //get float
                    System.out.println("PI is approximately equal to " + registers.getFloat32At(0));
                    System.out.println();

                //    m.disconnect();
                } catch (ModbusProtocolException e) {
                    e.printStackTrace();
                } catch (ModbusIOException e) {
                    e.printStackTrace();
                } catch (ModbusNumberException e) {
                    e.printStackTrace();
                }
                Thread.sleep(1000);
                }
     //       }
        } catch (RuntimeException e) {
            throw e;
        } catch (Exception e) {
            e.printStackTrace();
        }
    }
}

kochedykov, 2018-02-16:
Try using ReadInputRegisters.

gkrananth, 2018-02-16:
But it is holding register. Is it ok to use ReadInputRegisters?

kochedykov, 2018-02-16:
It is READ ONLY but holding register can be modified. So I guess that, actually, it is input register,

gkrananth, 2018-02-16:
anyhow i got a error Illegal Function

kochedykov, 2018-02-16:
I think you should ask the developer of the device you use because you send a valid request and device respond you with an exception. This issue does not relate to the library.

closed 2018-02-16 by kochedykov

gkrananth, 2018-03-26:
Hi
How to read holding register with reverse float 32 bit register value?

kochedykov, 2018-03-27:
Hello. Holding register takes 16 bit of memory and cannot store float32. If you mean that you have to read two registers holding one float32 and each register is a bit-reversed number then you can read these two registers and apply a bit-reversal algorithm for each of the two registers. To get a floating point number you have to combine the result by bitwise operations "left-shift" and "or".