kochedykov/jlibmodbus #4: Android things support

opengamer29, 2017-07-22:
Great library. Do you plan to add android things support? It could be extremely helpfull nowadays. Seems it isn't a big deal to add it. Android things have a pretty same interface for a serial port. https://developer.android.com/things/reference/com/google/android/things/pio/UartDevice.html

kochedykov, 2017-07-24:
Hello!

I did not plan that yet but if it really makes sense I'll add that feature. 

Thanks for your feedback!

opengamer29, 2017-07-25:
The star appeared right after a fairly quick response ;)

opengamer29, 2017-07-25:
I'm waiting for Android Things support with impatient. AT is well-known in IoT world now since this year. I have some AT projects and it would be great to use Modbus in them.

As I see it will not big a big deal: adding Bridge from AT UartDevice to your SerialPort interface and creating SerialPortATFactory.

kochedykov, 2017-07-25:
I can implement it immediately if you help me with debugging. Also it would be great if you give me a link to the AT library jar file[s] defining UartDevice.

opengamer29, 2017-07-25:
Yep, I'll help with debugging.

There are a lot of implementation of AT library based on device. For work with AT projects, you have to **provide** dependency from Gradle. There is a link to bintray - https://bintray.com/google/androidthings/androidthings-supportlibrary.
I don't know exactly how to get jar from this (if it is possible).

I'll try to implement this in my project by Inheritance from SerialPortAbstractFactory and implementing SerialPort interface. After that, I'll setSerialPortFactory to my new SerialPortFactory. Will it work? If yes, I'll do this and share results with you.

kochedykov, 2017-07-26:
>>Will it work?
Yes, you're right. Looking forward to your solution!

kochedykov, 2018-05-31:
Done :)

closed 2018-05-31 by kochedykov