kochedykov/jlibmodbus #17: Invalid checksum issue fixed for RTU frame is also present in ASCII framer

mlasevich, 2018-03-24:
The issue is same as https://github.com/kochedykov/jlibmodbus/issues/11, only for ASCII framer instead of RTU. If checksum happen to be 0, you get an exception raised 

```
Caused by: com.intelligt.modbus.jlibmodbus.exception.ModbusChecksumException: checksum error: received 0, calculated 0
	at com.intelligt.modbus.jlibmodbus.net.stream.InputStreamASCII.frameCheck(InputStreamASCII.java:60)
	at com.intelligt.modbus.jlibmodbus.net.transport.ModbusTransportSerial.read(ModbusTransportSerial.java:46)
	at com.intelligt.modbus.jlibmodbus.net.transport.ModbusTransport.readMessage(ModbusTransport.java:69)
	at com.intelligt.modbus.jlibmodbus.net.transport.ModbusTransport.readResponse(ModbusTransport.java:64)
	at com.intelligt.modbus.jlibmodbus.master.ModbusMaster.readResponse(ModbusMaster.java:111)
	at com.intelligt.modbus.jlibmodbus.master.ModbusMaster.processRequest(ModbusMaster.java:134)
	at com.intelligt.modbus.jlibmodbus.master.ModbusMaster.readHoldingRegisters(ModbusMaster.java:193)
```

kochedykov, 2018-03-26:
Thank you, fixed.

closed 2018-03-26 by kochedykov

mlasevich, 2018-03-28:
Thank you for a quick fix, is there any eta when this will make it to the maven repos? Seems like it is still 1.2.9.2 from December there. Thank you again.

mlasevich, 2018-03-28:
Actually, I just run against a build compiled from master branch and I am now seeing every single request fail. It seems it is calculating 0 in every scenario :-(

```
Caused by: com.intelligt.modbus.jlibmodbus.exception.ModbusChecksumException: checksum error: received -91, calculated 0
	at com.intelligt.modbus.jlibmodbus.net.stream.InputStreamASCII.frameCheck(InputStreamASCII.java:60)
	at com.intelligt.modbus.jlibmodbus.net.transport.ModbusTransportSerial.read(ModbusTransportSerial.java:46)
	at com.intelligt.modbus.jlibmodbus.net.transport.ModbusTransport.readMessage(ModbusTransport.java:69)
	at com.intelligt.modbus.jlibmodbus.net.transport.ModbusTransport.readResponse(ModbusTransport.java:64)
	at com.intelligt.modbus.jlibmodbus.master.ModbusMaster.readResponse(ModbusMaster.java:111)
	at com.intelligt.modbus.jlibmodbus.master.ModbusMaster.processRequest(ModbusMaster.java:134)
	at com.intelligt.modbus.jlibmodbus.master.ModbusMaster.readInputRegisters(ModbusMaster.java:214)
```

kochedykov, 2018-03-29:
Hi. I'm sorry, that is my fault. There were two issues: unimplemented write(byte[],int,int) in OutputStreamASCII, also invalid LRC calculation and reading sequence. I have fixed that and already updated the master branch. 
Today I am going to upload a new release to the maven central. Thank you!

commit 4d6b52f8 (2018-03-26): Fix #17: incorrect logic if lrc equals zero. thanks to https://github.com/mlasevich