kochedykov/jlibmodbus #18: Android as slave Bind Exception

bbetter, 2018-04-02:
getting this exception when setting up library on android. on listen() method
java.net.BindException: bind failed: EACCES (Permission denied)

Another thing is that in this lines
 ``` @Override
    synchronized public void listenImpl() throws ModbusIOException {
        try {
            server = new ServerSocket(tcp.getPort());
            mainThread = new Thread(this);
            setListening(true);
            mainThread.start();
        } catch (IOException e) {
            throw new ModbusIOException(e);
        }
    }
```
we only pass port to server socket , and not using any other data, it might be that I misunderstood your idea, but can you please explain it a bit, either here or in private message.

kochedykov, 2018-04-04:
> getting this exception
Have you granted the required permissions to your application?

> we only pass port to server socket , and not using any other data
It just uses the standard length of the queue and accepts connections on all local addresses.

bbetter, 2018-04-04:
yes, sorry, already fixed, i needed master not a slave actually, and it goes well on emulator.

kochedykov, 2018-04-04:
it's ok. if you have more questions, I'm around.

closed 2018-04-04 by kochedykov