kochedykov/jlibmodbus #19: ModbusChecksumException: checksum error arise again!

penggle, 2018-04-25:
Caused by: com.intelligt.modbus.jlibmodbus.exception.ModbusChecksumException: checksum error: received 59392, calculated 48617
	at com.intelligt.modbus.jlibmodbus.net.stream.InputStreamRTU.frameCheck(InputStreamRTU.java:44) ~[jlibmodbus-1.2.9.4.jar!/:?]
	at com.intelligt.modbus.jlibmodbus.net.transport.ModbusTransportSerial.read(ModbusTransportSerial.java:46) ~[jlibmodbus-1.2.9.4.jar!/:?]
	at com.intelligt.modbus.jlibmodbus.net.transport.ModbusTransport.readMessage(ModbusTransport.java:69) ~[jlibmodbus-1.2.9.4.jar!/:?]
	at com.intelligt.modbus.jlibmodbus.net.transport.ModbusTransport.readResponse(ModbusTransport.java:64) ~[jlibmodbus-1.2.9.4.jar!/:?]
	at com.intelligt.modbus.jlibmodbus.master.ModbusMaster.readResponse(ModbusMaster.java:111) ~[jlibmodbus-1.2.9.4.jar!/:?]
	at com.intelligt.modbus.jlibmodbus.master.ModbusMaster.processRequest(ModbusMaster.java:134) ~[jlibmodbus-1.2.9.4.jar!/:?]
	at com.intelligt.modbus.jlibmodbus.master.ModbusMaster.readHoldingRegisters(ModbusMaster.java:193) ~[jlibmodbus-1.2.9.4.jar!/:?]
	at com.certusnet.flexedge.modbus.service.ModbusExampleService.readHoldingRegisters(ModbusExampleService.java:65) ~[classes!/:1.0.0]

penggle, 2018-04-25:
Add: this error appears in version 1.2.9.4.

kochedykov, 2018-05-10:
Could show the output with debug mode on (Modbus.setLogLevel(Modbus.LogLevel.LEVEL_DEBUG))? There are no checksum errors for me while debugging using random numbers.

closed 2018-05-22 by kochedykov

gammal, 2019-08-23:
I'm getting the same error on the latest version (1.2.9.7).  How do I go about troubleshooting the problem, and what additional information do I have to share?  The stack trace of the exception I get doesn't contain more information than the one above.  I'm trying to read from a serial device connected through USB, and I'm actually able to read correctly from the device using different software.