kochedykov/jlibmodbus #22: ModBus Master TCP - Socket left open

AntosDR, 2018-06-01:
Hi,
I'm using your library and I have noticed that the `ModbusTransportTCP` class do not store each new created socket in any variable and therefore the sockets are never closed!
Analyzing the source code I found the way the socket is constructed and how it is passed to input and output streams. The problem occurs when the `ModbusTransportTCP.close()` method is invoked: only input and output streams are closed, but the socket is left opened.

At `ModbusTransportTCP`, row 40, it is possible to notice that the socket is not stored (it is stored in the input and output stream, but in any case it is left opened when the close method is invoked).
At `ModbusTransport`, row 45, it is possible to see how the close method is propagated only to input and output streams.

This is fine if you use only one connection to do everything but when a connection is opened and closed multiple times (before garbage collectors runs, of course) to the same ModBus endpoint, this situation happens:

[image]
I debugged my application and I'm sure that the `ModbusMaster.disconnect()` method is invoked each time, but the connections remains opened, as you can see in the screenshot. When this situation happens, the only solution is to wait that the connections are garbage collected and closed by the operating systems (or restarting your application, of course).

I'm using version 1.2.9.5 with Maven.

If you need more information please don't hesitate to ask.

kochedykov, 2018-06-04:
Hi! 
I didn't get this behavior on my machines. According to the documentation, the sockets should be closed. 

In ModbusTransportTCP constructor:

`super(new InputStreamTCP(socket), new OutputStreamTCP(socket));`

In InputStreamTCP:

`final private BufferedInputStream in = new BufferedInputStream(s.getInputStream());`

BufferedInputStream#close:

> Closes this input stream and releases any system resources associated with the stream.

`if (input != null) input.close();`

Socket#getInputStream() javadoc:

> Closing the returned java.io.InputStream will close the associated socket.

kochedykov, 2018-06-04:
I found that s.close does not get invoked. Fixed, today I'll publish a new release including the fix.
Thank you!

closed 2018-06-04 by kochedykov