kochedykov/jlibmodbus #23: this gives null pointer exception when i use lib 1.2.9 but it works fine when i use 1.2.8.4,

SanduDS, 2018-06-11:
/*Exception in thread "AWT-EventQueue-0" java.lang.NullPointerException
	at com.intelligt.modbus.jlibmodbus.serial.SerialUtils.createSerial(SerialUtils.java:63)
	at com.intelligt.modbus.jlibmodbus.master.ModbusMasterRTU.<init>(ModbusMasterRTU.java:34)
	at com.intelligt.modbus.jlibmodbus.master.ModbusMasterFactory.createModbusMasterRTU(ModbusMasterFactory.java:48)
	at powerMeterConnector.PMconnection.<init>(PMconnection.java:43)
*/
public class PMconnection {
    private static PMconnection plc;
    private ModbusMaster master;
    
    public PMconnection() {
         SerialParameters sp = new SerialParameters();
        //Modbus.setLogLevel(Modbus.LogLevel.LEVEL_DEBUG); reading response
       try {
             
            
            
                String[] port_list =new SerialPortList().getPortNames();
            
            if (port_list.length > 0) {
              //  System.out.println("Port: "+port_list[0]);
                //set the compatible parameters for the PLC
                sp.setDevice(port_list[0]);
                sp.setBaudRate(SerialPort.BaudRate.BAUD_RATE_9600);
                sp.setDataBits(8);
                sp.setParity(SerialPort.Parity.EVEN);
                sp.setStopBits(1);
                //if(master==null){ //singleton (not the correct)
                    master = ModbusMasterFactory.createModbusMasterRTU(sp);
              // }
                
            }
        } catch (RuntimeException e) {
            throw e;
        } catch (Exception e) {
            e.printStackTrace();
        }
        
    
    
    }
    public static PMconnection getInstance() {
        if(plc==null){
            plc=new PMconnection();
        }
        return plc;
    }
    public ModbusMaster getMaster(){
        return master;
    
    }
  
}

SanduDS, 2018-06-11:
please let me what I got wrong. and power meter (SMD 220 eastron )gives me integer register values but I couldn't convert it to real values. I m new to Modbus

kochedykov, 2018-06-13:
Hi! JLibmodbus supports several uart connectors like jssc, rxtx, jserialcomm etc. Before you create a master you should set SerialPortFactory, for instance jssc:
`SerialUtils.setSerialPortFactory(new SerialPortFactoryJSSC());`
If you need to read float values you can use the code below:

            ReadHoldingRegistersRequest request = new ReadHoldingRegistersRequest();
            //Or ReadInputRegistersRequest request = new ReadInputRegistersRequest();
            request.setServerAddress(slaveId);
            request.setStartAddress(offset);
            request.setQuantity(2);
            ReadHoldingRegistersResponse response = (ReadHoldingRegistersResponse) 
            request.getResponse();

            master.processRequest(request);
            ModbusHoldingRegisters registers = response.getHoldingRegisters();
            registers.getFloat32At(0);

closed 2018-06-14 by kochedykov