kochedykov/jlibmodbus #24: WARNING when build with maven

browncrane, 2018-06-22:
Find 
> [WARNING] The POM for com.intelligt.modbus:jlibmodbus:jar:1.2.9.7 is invalid, transitive dependencies (if any) will not be available, enable debug logging for more details

when building with maven.

Dependency in my `pom.xml` is

>         <dependency>
>             <groupId>com.intelligt.modbus</groupId>
>             <artifactId>jlibmodbus</artifactId>
>             <version>1.2.9.7</version>
>         </dependency>

jgrassini, 2018-06-25:
Here is the error message that causes the warning

```
[ERROR] 'dependencies.dependency.systemPath' for javax.comm:comm:jar 
must specify an absolute path but is ${project.basedir}/lib/comm.jar 
```

kochedykov, 2018-06-26:
Hi there.
I think, the best solution is to stop supporting javax.comm in the next release.

AntosDR, 2018-07-04:
Hi there!
I'm using this library in an OSGi context and I can confirm that building with Maven the javax.comm gives some bad issues but resolving OSGi dependencies is even worst! The problem is that this library (I copied the jar "comm.jar" from this repository) contains also an unaligned **example** folder which complicates things even more.
This different version resolved some issues when compiling as OSGi bundle, but I don't know if it contains everything jlibmodbus needs: http://www.java2s.com/Code/Jar/c/Downloadcomm20jar.htm
I agree with @kochedykov and @browncrane that stop supporting the javax.comm would be the best solution.

closed 2018-09-20 by kochedykov