kochedykov/jlibmodbus #27: ModbusMasterFactory.createModbusMasterRTU raise exeption

perrym, 2018-07-13:
Hello,
In Java is not strong :) Experience is no more than six months, so I had to learn Eclipse RCP and write a modbus driver.
I use the link of your library (downloaded fresh jlibmodbus-1.2.9.7.jar from sourceForge) with jssc.
I encountered the problem that when the ModbusMasterFactory.createModbusMasterRTU function is called  the nullptr ex is catched
Once (months 4-5 back) used the source code of your library - and this problem was not observed - the port was set correctly and everything worked. And now here I decided to use jar - and catch the ex (it's clear that for so many months the source code has changed, if I do not understand right away - then I'll debug with your updated sourses). Maybe you can tell me where to dig :)

Judging by the changes - added only thread safety.

[SEVERE] - 22:54:39.647  13-07-2018 - [ru.project.core.script.ScriptProcessor.connectMeasure] - java.lang.NullPointerException
	at ru.project.common.modbus.ModbusMasterPool.getModbusMaster(ModbusMasterPool.java:33)
	at ru.project.common.modbus.ModbusMasterHelper.<init>(ModbusMasterHelper.java:32)
	at ru.project.driver.measure.MeasureDriver.initMeasures(MeasureDriver.java:79)
	at ru.project.driver.measure.MeasureDriver.doStart(MeasureDriver.java:218)
	at ru.project.core.script.ScriptProcessor.connectMeasure(ScriptProcessor.java:211)
	at ru.project.core.script.ScriptProcessor.prepare(ScriptProcessor.java:279)
	at ru.project.e4.parts.script.ScriptView$StartScriptJob.run(ScriptView.java:904)
	at org.eclipse.jface.operation.ModalContext$ModalContextThread.run(ModalContext.java:119)

import java.util.HashMap;
import java.util.Map;

import com.intelligt.modbus.jlibmodbus.master.ModbusMaster;
import com.intelligt.modbus.jlibmodbus.master.ModbusMasterFactory;
import RS232Configuration;

public class ModbusMasterPool {

	private static Map<String, ModbusMaster> modbusPool = new HashMap<String, ModbusMaster>();

	public static ModbusMaster getModbusMaster(RS232Configuration cfg) throws Exception {
		ModbusMaster mm=null;
		try {

			if (!modbusPool.containsKey(cfg.getPort())) {
				mm = ModbusMasterFactory.createModbusMasterRTU(cfg.asSerialParameters());
				modbusPool.put(cfg.getPort(), mm);
			} else {
				mm = modbusPool.get(cfg.getPort());
			}

			if (!mm.isConnected()) {
				mm.setResponseTimeout(cfg.getTimeOut());
				mm.connect();
			}
		} 
		catch (Exception ex) {
			
			throw new Exception("Unable to connect to com port", ex);
		}
		return mm;
	}
	public static void disconnectAll() throws Exception {
		for (String port : modbusPool.keySet())
			if (modbusPool.get(port).isConnected())
				modbusPool.get(port).disconnect();
		modbusPool.clear();
	}

`}`

import com.intelligt.modbus.jlibmodbus.serial.SerialParameters;
import com.intelligt.modbus.jlibmodbus.serial.SerialPort.Parity;
import com.intelligt.modbus.jlibmodbus.serial.SerialPort.BaudRate;

public class RS232Configuration {
	private String port = "COM3";
	private BaudRate baudrate = BaudRate.BAUD_RATE_9600;
	private Parity parity = Parity.NONE;
	private int dataBits = 8;
	private int stopBits = 1;
	private int timeOut = 100;//

	public String getPort() {
		return port;
	}

	public void setPort(String port) {
		this.port = port;
	}

	public BaudRate getBaudrate() {
		return baudrate;
	}

	public void setBaudrate(BaudRate br) {
		this.baudrate = br;
	}

	public Parity getParity() {
		return parity;
	}

	public void setParity(Parity parity) {
		this.parity = parity;
	}

	public int getDataBits() {
		return dataBits;
	}

	public void setDataBits(int dataBits) {
		this.dataBits = dataBits;
	}

	public int getStopBits() {
		return stopBits;
	}

	public void setStopBits(int stopBits) {
		this.stopBits = stopBits;
	}

	public SerialParameters asSerialParameters() {
		SerialParameters serialParameters = new SerialParameters();
		serialParameters.setDevice(getPort());
		serialParameters.setBaudRate(getBaudrate());
		serialParameters.setParity(getParity());
		serialParameters.setDataBits(getDataBits());
		serialParameters.setStopBits(getStopBits());
		return serialParameters;
	}

	public void setTimeOut(int timeOut) {
		this.timeOut = timeOut;
	}

	public int getTimeOut() {
		// TODO Auto-generated method stub
		return timeOut;
	}
}

Thanks,
perrym

kochedykov, 2018-07-15:
Hi! You should set a serial port factory. Look at example ExampleRTU.java.
For instace, if you use jssc invoke `SerialUtils.setSerialPortFactory(new SerialPortFactoryJSSC());`

kochedykov, 2018-07-15:
I am going to add an automatic selection of a serial port factory in the next release

perrym, 2018-07-16:
Thansk a lot! All works!

closed 2018-07-16 by perrym