kochedykov/jlibmodbus #30: Implements Modbus Function 20(0x14)

apenzhulyak, 2018-10-01:
It would be great to show in simple Example how we can implements Modbus Function 20(0x14). I don`t understand how it`s works and always get TimeOutException but ModbusMaster.isConected

apenzhulyak, 2018-10-01:
What is wrong?              

              TcpParameters tcpParameters = new TcpParameters();
               try {
                  tcpParameters.setHost(InetAddress.getAllByName(Prefs.getInstance(getActivity()).getIpConfig())[0]);
               } catch (UnknownHostException e) {
                  e.printStackTrace();
               }
               tcpParameters.setPort(Modbus.TCP_PORT);
               tcpParameters.setKeepAlive(true);
               ModbusMaster m = ModbusMasterFactory.createModbusMasterTCP(tcpParameters);
               m.setResponseTimeout(5000);
               Modbus.setAutoIncrementTransactionId(true);
                
               try {
                 if (!m.isConnected()) {
                    m.connect();
                 }
               ModbusOutputStream mos = new ModbusOutputStream(){
                     @Override
                     public void write(int b) throws IOException {
                         super.write(b);
                     }
                     @Override
                     public void writeShortBE(int s) throws IOException {
                         super.writeShortBE(s);
                     }
               };
               ReadFileRecordRequest request = new ReadFileRecordRequest();
               m.processRequest(request);
               request.writeRequest(mos);
               ReadFileRecordResponse response = (ReadFileRecordResponse) request.getResponse();
               ModbusFileRecord[] fileRecords = response.getFileRecords();
               for (ModbusFileRecord file : fileRecords) {
                   Log.e(, "file: ", Arrays.toString(file.getRegisters()));
               }
               } catch (Exception ex) {
                    Log.e(, "exception: ", ex);
               }

kochedykov, 2018-10-24:
Hi! Could you show me the logs?

kochedykov, 2019-02-14:
Cannot help without feedback.

closed 2019-02-14 by kochedykov