kochedykov/jlibmodbus #32: Dependencies issues with Android things support that affect PC based projects

gopalakrishnan-subramani, 2018-10-26:
Two Issues

1. Version 1.2.9.7 breaks SBT build, I need to manually add below for SBT to download the package
   "Android Google  Maven2 Repository" at "https://maven.google.com/"
    
    I don't need Android, I build for Windows and Linux and Mac.

2. After adding Android Google Maven repo, it downloads android thing jars, but the program crash at runtime due to poor selection of Android library on Windows.

I see https://github.com/kochedykov/jlibmodbus/blob/master/src/com/intelligt/modbus/jlibmodbus/serial/SerialUtils.java source code,   

```java
static private Set<ValidatorSerialPortFactory> validatorSet = new TreeSet<ValidatorSerialPortFactory>();

    static {
        registerSerialPortFactory("com.google.android.things.AndroidThings", SerialPortFactoryAT.class.getCanonicalName());
        registerSerialPortFactory("com.fazecast.jSerialComm.SerialPort", SerialPortFactoryJSerialComm.class.getCanonicalName());
        registerSerialPortFactory("jssc.SerialPort", SerialPortFactoryJSSC.class.getCanonicalName());
        registerSerialPortFactory("purejavacomm.PureJavaComm", SerialPortFactoryPJC.class.getCanonicalName());
        registerSerialPortFactory("gnu.io.RXTXVersion", SerialPortFactoryRXTX.class.getCanonicalName());
    }
```

Can you provide a public method where we could remove not wanted registerSerialPortFactory?

kochedykov, 2018-10-26:
Please use 1.2.9.7 (commit c315832a5172fe90d2562cd823f701b810452d59). The next release is coming very soon, but now the stable version is 1.2.9.7.

kochedykov, 2019-02-14:
Fixed. Now I do not use .class. Just do not use trySelectConnector.

commit 83219ea0 (2019-02-14): Fix #32.
closed 2019-02-14 by kochedykov