kochedykov/jlibmodbus #33: Setting negative Integer in ModbusHoldingRegisters causes IllegalDataValueException

ippodamia, 2018-11-13:
I have a Modbus Slave that uses a DataHolder which has a number of  HoldingRegisters.
When i try to set a negative integer value with the function:
`this.slave.getDataHolder().getHoldingRegisters().set(offset, intValue);`
the ModbusHoldingRegisters.checkValue() method throws an IllegalDataValueException.

I think that the Modbus.checkRegisterValue() method should check for a range with min value=-32768 (minimum signed 16-bit integer value) instead of 0.

browncrane, 2018-11-30:
Could you paste more related code? Did you consider using `short`?

ippodamia, 2018-12-03:
Here is a simple example:
`DataHolder dh = new SimpleDataHolderBuilder(5).build();`
`dh.getHoldingRegisters().set(0, -1);`
that causes the following exception:
`com.intelligt.modbus.jlibmodbus.exception.IllegalDataValueException: ILLEGAL_DATA_VALUE: Exception Code = 3 at com.intelligt.modbus.jlibmodbus.data.ModbusHoldingRegisters.checkValue (ModbusHoldingRegisters .java :197) `

I now solved the problem by writing the negative value as the unsigned 16-bit value e.g.
`int signedInt32 = -1;`
`int unsignedInt16 = signedInt32 & 0xFFFF; // 65535`
`dh.getHoldingRegisters().set(0, unsignedInt16);`

When reading out the value I cast it into a short in order to get the signed value e.g.
`int readUnsignedInt16 = dh.getHoldingRegisters().get(0); // 65535`
`int readSignedInt32 = (short)readUnsignedInt16; // -1`

kochedykov, 2018-12-07:
> @kochedykov Would `short` type parameter supporting be a good idea?
@browncrane I think the parameter should be cast to a short value inside the "setImpl" method body.

kochedykov, 2018-12-07:
@browncrane Something like that:
`checkValue((value.shortValue())&0xffff);`

kochedykov, 2018-12-07:
@ippodamia 
We'll fix it.

> Could you paste more related code? Did you consider using `short`?

ippodamia, 2018-12-12:
Great! Thank you for the good work!

commit 38fcf02f (2019-02-14): Fix #33.
closed 2019-02-14 by kochedykov