kochedykov/jlibmodbus #34: Wrong implementation at the DataUtils.dwordFromWords

du4, 2018-11-21:
Hello.
Your implementation looks like
`public static int dwordFromWords(int l, int h) {
        return (l & 0xffff) & ((h & 0xffff) << 16);
    }`
There is mistake at the AND operation between **_l_** and **_h_** at the return statement. I think you should use OR operation.

browncrane, 2018-11-23:
The method is never used. It just reminds me Modbus Holding register(as the word in this domain) is 16bits and the `int` in Java is 32 bits. It could cause a problem when dealing with a negative value. e.g. 
-3 as a 16bit signed value is `0xFFFD`. After storing it with `int` in Java, it becomes `0x0000FFFD`
the sign bit is changed to 0, and so it would be interpreted as 65533.

browncrane, 2018-11-24:
@du4 if `h` only contains the value in lower 16bits, the code is correct.

du4, 2018-11-24:
In your implementation, the result will always be zero because you use AND operation between hi<<16 and 
 low parts of java integer.
PS: when you working with bits it doesn’t depend on which numbers you use signed or unsigned

kochedykov, 2018-11-30:
> In your implementation, the result will always be zero because you use AND operation between hi<<16 and
> low parts of Java integer.
> PS: when you working with bits it doesn’t depend on which numbers you use signed or unsigned

Certainly, it will be fixed in the next release.

commit ad405adc (2019-02-14): Fix #34.
closed 2019-02-14 by kochedykov