kochedykov/jlibmodbus #52: Register value out of range : 65280

Andrii10, 2020-06-17:
Hello I have problem with last version jlibmodbus 1.2.9.7
when I use function x05 WriteSingleRegister
in
_public void setValue(int value) throws ModbusNumberException {
        if (!Modbus.checkRegisterValue(value)) {
            throw new ModbusNumberException("Register value out of range", value);
        }
        this.value = ((short) value) & 0xffff;
    }_ 
I take error  Register value out of range : 65280
in method
_static public boolean checkRegisterValue(int value) {
        return checkRange((short)value, 0, Modbus.MAX_REGISTER_VALUE);
    }_

And I fix the checkRange(_value,_ 0, Modbus.MAX_REGISTER_VALUE) after that it worked for me
Did this right or not help me?

FutureGUIs, 2020-06-17:
I just had to do the same thing, remove the 'short' cast.  and so far it's working fine for me.

jonasgerne, 2020-11-26:
Having the same issue, using writeMultipleCoils as a workaround...

kochedykov, 2021-02-03:
Yes. It is a bug. Please make a pull request.