kochedykov/jlibmodbus #53: jserialcomm 2.6.2 problem

rdkartono, 2020-07-07:
on jserialcomm 2.6.2, have problem with SetComPortParameter.

My java code 
`/**
	 * Open Serial Port for communication
	 * @param serialname : serial port name
	 * @param baudrate : baud rate, typically 4800, 9600, 144400, 19200, 38400, 57600, 115200
	 * @param databit : data bit length , typically 8
	 * @param stopbit : typically 1
	 * @param parity : 0 = None, 1 = Odd, 2 = Even, 3 = Mark, 4 = Space
	 * @return true if success
	 */
	public boolean Open(String serialname, int baudrate, int databit, int stopbit, int parity) {
		sp = new SerialParameters();
		
		sp.setDataBits(databit);
		sp.setDevice(serialname);
		sp.setStopBits(stopbit);
		sp.setBaudRate(BaudRate.getBaudRate(baudrate));
		sp.setParity(Parity.getParity(parity));
		
		SerialUtils.setSerialPortFactoryJSerialComm();
		
		try {
			
			modbus = ModbusSlaveFactory.createModbusSlaveRTU(sp);
			
			modbus.addListener(listener);
			
			modbus.listen();
			return true;
		} catch (SerialPortException e) {
			raise_log("Open failed, Msg : "+e.getMessage());
			return false;
		} catch (ModbusIOException e) {
			raise_log("Modbus Listen failed, Msg : "+e.getMessage());
			return false;
		}
	}`

Log :
`java.lang.NoSuchMethodError: com.fazecast.jSerialComm.SerialPort.setComPortParameters(IIII)V
	at com.intelligt.modbus.jlibmodbus.serial.SerialPortJSerialComm.open(SerialPortJSerialComm.java:71)
	at com.intelligt.modbus.jlibmodbus.net.ModbusConnectionSerial.openImpl(ModbusConnectionSerial.java:60)
	at com.intelligt.modbus.jlibmodbus.net.ModbusConnection.open(ModbusConnection.java:45)
	at com.intelligt.modbus.jlibmodbus.slave.RequestHandlerSerial.run(RequestHandlerSerial.java:51)
	at java.lang.Thread.run(Thread.java:748)`

JardarM, 2021-01-13:
From what I understand there is only a rebuild required and updated reference to jseriallib 2.6.2 to make new release. Would be highly appreciated if somebody could do this?

kochedykov, 2021-02-03:
The new release is coming soon.

pfried, 2021-07-15:
@kochedykov I know it is a bit annoying if people request things. But this release would really save me some real trouble since we are now somehow stuck with a inferior jserialcomm when using jlibmodbus. Could you make a new release?

emilm, 2022-06-08:
> The new release is coming soon.

@kochedykov  any news on this? :)

kochedykov, 2022-11-15:
it's still alive!)

kochedykov, 2022-11-15:
> I guess this project is dead.

it's still alive!)

ramen2k10, 2022-12-07:
Serial Communication support both stopbit 1 and 2 then why in the above code in parameters details it mentioned - @param stopbit : typically 1 ?? Is jSerialComm supports only stop bits 1 ?