kochedykov/jlibmodbus #54: Next Release?

antiquechrono, 2020-08-12:
There hasn't been a release since 2018 but there are changes on master that fix some bugs. Most notably the pom is broken and the ModbusCoils class doesn't actually return the coils. It looks like both of these issues have been corrected but no release was ever created. Any chance of a new release so these issues get fixed?

shdk, 2020-10-06:
I'm interested too.

shdk, 2020-10-06:
I'm interested too. @kochedykov

kochedykov, 2021-02-03:
The new release is coming soon. I was busy with another large project.

shdk, 2021-02-03:
> 
> 
> The new release is coming soon. I was busy with another large project.

Thanks @kochedykov. I am really looking forward for it.

leoandrea7, 2021-05-14:
Hi @kochedykov, any news about next release?

yongseokleecom, 2021-06-16:
I'm also waiting for the release ~!

ajb113, 2021-09-20:
Hi @kockedykov thanks for providing this library.

Are you still maintaining this project? I'm running into the issue that was fixed in https://github.com/kochedykov/jlibmodbus/commit/63d142928f9dc61f496f19df83c97a224f734145 so am also interested in a new release.

AlexGuironnetRTE, 2021-11-25:
Hello,

I'd be interested too, issue https://github.com/kochedykov/jlibmodbus/issues/24 means we have to add the jar with the fix version to our repo as our Gradle build fails with the current release.
Is there anything we can do to help with the release ?

Thanks for providing this library !

shdk, 2023-01-21:
@kochedykov, can you release a new version including the last commits, please?

kochedykov, 2024-04-18:
Done!

closed 2024-04-18 by kochedykov