kochedykov/jlibmodbus #74: com.intelligt.modbus.jlibmodbus.msg.response.BroadcastResponse cannot be cast to com.intelligt.modbus.jlibmodbus.msg.response.ReadHoldingRegistersResponse

shaisxx, 2022-04-18:
com.intelligt.modbus.jlibmodbus.msg.response.BroadcastResponse cannot be cast to com.intelligt.modbus.jlibmodbus.msg.response.ReadHoldingRegistersResponse

yudaichen, 2022-09-07:
java.lang.ClassCastException: com.intelligt.modbus.jlibmodbus.msg.response.BroadcastResponse cannot be cast to com.intelligt.modbus.jlibmodbus.msg.response.ReadInputRegistersResponse

yudaichen, 2022-09-08:
Cannot be a broadcast address.    <pre> request.getServerAddress() != Modbus.BROADCAST_ID </pre> ServerAddress != 0x00.  It took me a whole day.  I once thought it was the problem of spring-boot-devtools.

Hotant-Bu, 2023-08-08:
cannot be cast to com.intelligt.modbus.jlibmodbus.msg.response.ReadHoldingRegistersResponse