kochedykov/jlibmodbus #78: Cannot create more than one slave under a TCP connection

Fhui, 2022-07-16:
I want to create multiple slaves after establishing a TCP connection, but the slave object is bound to the parameters of the TCP connection. If you continue to create it, there will be an exception that the port has been occupied, so is there any other way?

Fhui, 2022-07-16:
I don't want to establish multiple TCP connections. I just want to create multiple slaveids after establishing a TCP connection

mastera74, 2022-08-10:
Did you manage to resolve this issue?

halimselim, 2022-09-12:
I developed a dirty solution for myself. I forwarded slaveid to dataholder.
Disabled slave id filter in [RequestHandlerTCP.java] to catch all request

If you like my solution I would create a pull request.

https://github.com/kochedykov/jlibmodbus/compare/master...halimselim:jlibmodbus:master

```
public class MyDataHolder extends DataHolder {

    @Override
    public int[] readHoldingRegisterRange(int slaveid, int offset, int quantity) throws IllegalDataAddressException {

    }
}
```

Fhui, 2022-11-08:
thinks

Fhui, 2022-11-08:
> 我为自己开发了一个肮脏的解决方案。我将 slaveid 转发给数据持有者。 在 [RequestHandlerTCP.java] 中禁用 slave id 过滤器以捕获所有请求
> 
> 如果您喜欢我的解决方案，我会创建一个拉取请求。
> 
> [主... halimselim:jlibmodbus:master](https://github.com/kochedykov/jlibmodbus/compare/master...halimselim:jlibmodbus:master)
> 
> ```
> public class MyDataHolder extends DataHolder {
> 
>     @Override
>     public int[] readHoldingRegisterRange(int slaveid, int offset, int quantity) throws IllegalDataAddressException {
> 
>     }
> }
> ```
thinks 

I'm going to use another library, modbus4j, it's good

Fhui, 2022-11-08:
> 我为自己开发了一个肮脏的解决方案。 在 [RequestHandler.java] 中禁止slaveid过滤器以请求所有请求
> 
> 如果您喜欢我的解决方案，我会创建一个拉取请求。
> 
> [主... halimselim:jlibmodbus:master](https://github.com/kochedykov/jlibmodbus/compare/master...halimselim:jlibmodbus:master)
> 
> ```
> public class MyDataHolder extends DataHolder {
> 
>     @Override
>     public int[] readHoldingRegisterRange(int slaveid, int offset, int quantity) throws IllegalDataAddressException {
> 
>     }
> }
> ```

thanks

I'm going to use another library, modbus4j, it's good