kochedykov/jlibmodbus #93: Error when connecting to Modbus RTU and connecting with jSerialComm

kk99999, 2024-01-21:
I am creating Modbus RTU and trying to connect using jSerialComm. I am getting below error though the method exists in jSerialComm.

java.lang.NoSuchMethodError: 'void com.fazecast.jSerialComm.SerialPort.setComPortParameters(int, int, int, int)'

I verified that there is only one version of jSerialComm(2.10.4) library. Below is the code to connect.
SerialParameters serialParameters = new SerialParameters("COM1", SerialPort.BaudRate.BAUD_RATE_9600, 8, 1, SerialPort.Parity.EVEN);
SerialUtils.setSerialPortFactoryJSerialComm();
ModbusMasterFactory.createModbusMasterRTU().connect();

Any thoughts?

kk99999, 2024-01-25:
I am seeing this issue with jSerialComm versions  > 2.6.0. If using 2.6.0 and earlier version I don't see this issue. Any chance this can be verified and may be resolved?

kochedykov, 2024-02-06:
Just testet with JSerialComm v2.10.4. Everything is OK.

divjad---, 2024-03-14:
@kochedykov Hi, for me this error still persists. I am using jLibModbus v1.2.9.9 and jSerialComm v2.10.4.

kochedykov, 2024-03-16:
Hi! Show me the error log.

kochedykov, 2024-03-30:
[image]

Did you add the jSerialComm path to classpath?

> Yes, here is the error log:
> 
> `Caused by: java.lang.NoSuchMethodError: 'void com.fazecast.jSerialComm.SerialPort.setComPortParameters(int, int, int, int)' at com.intelligt.modbus.jlibmodbus.serial.SerialPortJSerialComm.open(SerialPortJSerialComm.java:71) at com.intelligt.modbus.jlibmodbus.net.ModbusConnectionSerial.openImpl(ModbusConnectionSerial.java:60) at com.intelligt.modbus.jlibmodbus.net.ModbusConnection.open(ModbusConnection.java:45) at com.intelligt.modbus.jlibmodbus.master.ModbusMaster.connectImpl(ModbusMaster.java:76) at com.intelligt.modbus.jlibmodbus.master.ModbusMaster.connect(ModbusMaster.java:90) *...omitted* at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method) at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(Unknown Source) at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source) at java.base/java.lang.reflect.Method.invoke(Unknown Source) at org.jboss.weld.injection.StaticMethodInjectionPoint.invoke(StaticMethodInjectionPoint.java:95) ... 45 more`

divjad---, 2024-04-01:
Yes, it is added to the classpath, library is also found inside the packed JAR.

divjad---, 2024-04-15:
Hi, just a note. This issue seems to be resolved when updating jSerialComm to the latest version inside jlibmodbus. And also jlibmodbus version 1.2.9.11 from Maven Central seems to resolve this issue.

kochedykov, 2024-04-18:
Ok, closed!

closed 2024-04-18 by kochedykov