kochedykov/jlibmodbus #97: ModbusNumberException when writing a "true" into a coil

chk2902, 2024-08-21:
`com.intelligt.modbus.jlibmodbus.exception.ModbusNumberException: Register value out of range : 65280
	at com.intelligt.modbus.jlibmodbus.msg.request.WriteSingleRegisterRequest.setValue(WriteSingleRegisterRequest.java:103)
	at com.intelligt.modbus.jlibmodbus.msg.request.WriteSingleCoilRequest.setCoil(WriteSingleCoilRequest.java:71)
	at com.intelligt.modbus.jlibmodbus.msg.ModbusRequestBuilder.buildWriteSingleCoil(ModbusRequestBuilder.java:93)
	at com.intelligt.modbus.jlibmodbus.master.ModbusMaster.writeSingleCoil(ModbusMaster.java:285)`

This happens as the "true" coil value is 0xff00 and is checked to be in the range of 0 to 0xffff, but the range check in jave seems a bit unintuitive 

`checkRange((short)value, 0, Modbus.MAX_REGISTER_VALUE)`

which will convert the value to a short (-32768..32767) and thus it is a negative value, not being in the int range!

chk2902, 2024-08-21:
Ah, I see I have 1.2.9.7 which is outdated. Sorry, I guess it's fixed (Cannot check here, the failed check happens at a customer)

chk2902, 2024-08-21:
BTW: automatically adding a dependency to the new version 1.2.9.9 does not work in Android Developer Studio, here the most current version is 1.2.9.7.

chk2902, 2024-08-21:
... and your last build number seems to be "<version>1.2.9.10</version>"

kochedykov, 2024-09-19:
> ... and your last build number seems to be "1.2.9.10"

Hi! I've just updated the version info. The latest release is 1.2.9.11!

The latest stable release.
<dependency>
  <groupId>com.intelligt.modbus</groupId>
  <artifactId>jlibmodbus</artifactId>
  <version>1.2.9.11</version>
</dependency>